egony/cc2652p_E72-2G4M20S1E #1: Cant get into bootload anymore

sintei, 2020-10-23:
Hello,

I use SLAESH CC2652R stick (info here https://slae.sh/projects/cc2652/)
I flashed it with your firmware CC2652P_E72_20200921.zip
After this, I cant get into bootloader anymore.
Normally, i press the bootloader button, and then reset then release boot loader button.
But SmartRF Flash program will not connect.

Error message:
>Initiate access to target: COM3 using 2-pin cJTAG.
>No response from device. Device may not be in bootloader mode. Reset device and try again.
If problem persists, check connection and baud rate.
>Connecting over serial bootloader failed: No response from device. Device may not be in bootloader mode. Reset device and try again.
If problem persists, check connection and baud rate.
>Failed to create device object.

The led is RED which indicates it is on (?).

Any advice?

egony, 2020-11-01:
Hello!
My firmware used DIO15 as bootloader activation pin. I think your stick used another pin for this button. So, you need to short DIO15 to GND instead bootloader button. I think :/ Another suggestion - use some hardware programmer like segger j-link.

sintei, 2020-11-01:
You are correct. I tried shorting GPIO15 instead and it worked. Thanks!

closed 2020-11-01 by sintei