egony/cc2652p_E72-2G4M20S1E #3: ZigBee stick v4 - external antenna modification

znanev, 2020-12-15:
Hello @egony,

and thanks for your great work brining [ZigBee stick v4](https://github.com/egony/cc2652p_E72-2G4M20S1E/wiki) to life!

This weekend I managed to build the PCB, but I have some questions about modifying the module to support external antenna. Here's a snapshot from the picture on your Wiki page:

https://github.com/egony/cc2652p_E72-2G4M20S1E/wiki
[image]

From what I could see at that picture, I've cut the tracks near the pins of the SMA connector, which I circled in red ink. But when I powered on the board, the LDO got really hot, so I disconnected the power supply. Measuring the resistance between the pins of the SMA connector, it appears that they are all shorted. So my question is - is the track, circled in yellow, also cut? I can't judge from the photo alone if it is just scratched or it is cut out.

Many thanks!

egony, 2020-12-15:
Hello!

Now I added picture on Wiki, please, take a look on it.
And no, you shouldn't cut track circled in yellow. Zero resistance - it's ok.

znanev, 2020-12-15:
Thanks for your quick response!

It's good that I asked, I was about the cut out the yellow track - it seems that the opposite one should be cut out instead 👍 

I double and triple checked the PCB for shorts, but nothing obvious. But if there was a true short circuit, the output voltage of the LDO should be close to 0? When I measured it, it was about 3.2V, which is within specs.

I'm not sure if it means anything, but I haven't soldered the CH340 yet (waiting for these ICs to arrive), and also haven't flashed your firmware into the module - it is with what came from the factory...

egony, 2020-12-15:
I never tests this condition - w/o uart and with another firmware. Any way, LDO must stay cold. When it too hot - it is bad. Unsolder parts one by one and check (begin with 22uf capacistor).

znanev, 2020-12-15:
Just checked all passive components terminals and I've found no shorts.

Strange thing is, I've now plugged the board into my laptop's USB port (USB 3, so should be able to deliver 900mA) and the LDO got warm, but not hot. Maybe I was alarmed yesterday, because I compared it with a v3 board (rev 1.7), which LDO was not that warm. It might be that the E72 module consumes significantly more power than the other, cc2538+cc2592 based module.

But I'll know for sure once/if I manage to program the right firmware in it.

znanev, 2021-01-01:
Happy New Year!

The CH340 chips arrived yesterday. Today I soldered the missing IC and after flashing your firmware via JTAG, the device appears to be functioning. The LDO is definitely cold when touched, so the warm condition that I experienced earlier might be related to the factory-programmed firmware of the E72 module (or lack of any firmware, I'm not sure how these come out of the factory).

So all is good now, thanks for your guidance, @egony!

closed 2021-01-04 by znanev