egony/cc2652p_E72-2G4M20S1E #8: Ошибка 0x02: INVALID_PARAM

andrii-kulikov-epam, 2021-02-02:
Плата: "Плата в форм-факторе стика v3"
Модуль: E72 2G4M20S1E
SMA разъем: Используется

Салют!
Столкнулся с проблемой
Спустя 12-24 часа работы стика все устройства в сети Z2m перестают работать
Перезагрузка плагина не помогает, ругается на то что стик недоступен
Последняя ошибка в логах:
```
Zigbee2MQTT:info  2021-01-31 21:10:18: MQTT publish: topic 'zigbee2mqtt/[Hallway] Wall switch', payload '{"linkquality":81,"state":"ON","state_left":"ON","state_right":"ON","temperature":29}'
Zigbee2MQTT:error 2021-01-31 21:27:51: Publish 'set' 'state' to '[Hallway] Wall switch' failed: 'Error: Command 0x00158d0003464047/3 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SREQ '--> AF - dataRequest - {"dstaddr":7472,"destendpoint":3,"srcendpoint":1,"clusterid":6,"transid":106,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,135,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'
Zigbee2MQTT:info  2021-01-31 21:27:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '[Hallway] Wall switch' failed: 'Error: Command 0x00158d0003464047/3 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (SREQ '--> AF - dataRequest - {\"dstaddr\":7472,\"destendpoint\":3,\"srcendpoint\":1,\"clusterid\":6,\"transid\":106,\"options\":0,\"radius\":30,\"len\":3,\"data\":{\"type\":\"Buffer\",\"data\":[1,135,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'","meta":{"friendly_name":"[Hallway] Wall switch"},"type":"zigbee_publish_error"}'
```

В настройках Z2m **не** указывал
```yaml
advanced:
  baudrate: 115200
  rtscts: false
```

Помогите пожалуйста решить проблему, очень уж хочу с Conbee 2 переехать :)

egony, 2021-02-02:
Много девайсов в сети? 
Репортинг используется?
Как восстанавливаете?

andrii-kulikov-epam, 2021-02-02:
- Девайсов в сети около 20
- Репортинг это что такое?
- Чтоб восстановить работу сети надо в ручную вынять/вставить стик и перезапустить плагин Z2m

Update:
- Обновил прошилвку до "[CC1352P2_CC2652P_other_20210120.zip](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin)", так как нашел в Issues рекомендацию от **Koenkk** что можно стики **CC2652P**  шить прошивками от **CC1352P**
- Настроил маленький стенд для тестирования. Подключил [розетку Xiaomi](https://www.zigbee2mqtt.io/devices/ZNCZ02LM.html) и [кнопку Ikea](https://www.zigbee2mqtt.io/devices/E1524_E1810.html)
- Спустя пару часов я попробовал из UI Z2m переключить состояние розетки, на что получил сообщение с похожей ошибкой

```
Zigbee2MQTT:info  2021-02-02 14:05:01: MQTT publish: topic 'zigbee2mqtt/0x00158d0001104c10', payload '{"consumption":682.01,"energy":682.01,"linkquality":171,"power":0,"state":"ON","temperature":37}'
Zigbee2MQTT:info  2021-02-02 14:10:38: MQTT publish: topic 'zigbee2mqtt/0x00158d0001104c10', payload '{"consumption":682.01,"energy":682.01,"linkquality":168,"power":0,"state":"ON","temperature":37}'
Zigbee2MQTT:error 2021-02-02 14:15:59: Publish 'set' 'state' to '0x00158d0001104c10' failed: 'Error: Command 0x00158d0001104c10/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SREQ '--> AF - dataRequest - {"dstaddr":541,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":196,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,56,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'
Zigbee2MQTT:info  2021-02-02 14:15:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x00158d0001104c10' failed: 'Error: Command 0x00158d0001104c10/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (SREQ '--> AF - dataRequest - {\"dstaddr\":541,\"destendpoint\":1,\"srcendpoint\":1,\"clusterid\":6,\"transid\":196,\"options\":0,\"radius\":30,\"len\":3,\"data\":{\"type\":\"Buffer\",\"data\":[1,56,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'","meta":{"friendly_name":"0x00158d0001104c10"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:error 2021-02-02 14:16:03: Publish 'set' 'state' to '0x00158d0001104c10' failed: 'Error: Command 0x00158d0001104c10/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SREQ '--> AF - dataRequest - {"dstaddr":541,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":197,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,57,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'
Zigbee2MQTT:info  2021-02-02 14:16:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x00158d0001104c10' failed: 'Error: Command 0x00158d0001104c10/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (SREQ '--> AF - dataRequest - {\"dstaddr\":541,\"destendpoint\":1,\"srcendpoint\":1,\"clusterid\":6,\"transid\":197,\"options\":0,\"radius\":30,\"len\":3,\"data\":{\"type\":\"Buffer\",\"data\":[1,57,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'","meta":{"friendly_name":"0x00158d0001104c10"},"type":"zigbee_publish_error"}'
(node:25647) UnhandledPromiseRejectionWarning: Error: SREQ '--> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}' failed with status '(0xc2: NWK_INVALID_REQUEST)' (expected '(0x00: SUCCESS)')
    at Znp.<anonymous> (/Users/test/IdeaProjects/zigbee2mqtt-test/zigbee2mqtt-master/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
    at Generator.next (<anonymous>)
    at fulfilled (/Users/test/IdeaProjects/zigbee2mqtt-test/zigbee2mqtt-master/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:25647) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:25647) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

egony, 2021-02-04:
Возможно, проблема в чипе ch340 или в настройках энергосбережения хоста.
Была одна проблема в SDK, но ее, вроде, пофиксили в свежих прошивках:
https://github.com/Koenkk/zigbee2mqtt/discussions/6019

andrii-kulikov-epam, 2021-02-14:
Перепаял USB-UART преобразователь, стик работает уже сутки без ошибок
Спасибо большое за то что помог решить проблему 🙂

closed 2021-02-14 by andrii-kulikov-epam