egony/cc2652p_E72-2G4M20S1E #12: Uart E72 модуля

wisesokol, 2021-02-27:


wisesokol, 2021-02-27:
Добрый день. Собираюсь шить модуль через uart c помощью стандартного bootloader и активации backdoor (DIO_13 to high на самом 2652 чипе). Не dio_11. [https://e2e.ti.com/support/wireless-connectivity/bluetooth/f/538/t/871984?LAUNCHXL-CC26X2R1-Bootloader-backdoor-enable-pin](url)
Вопрос:
Не выведены ли dio_2 и dio_3 из под колпака. Может кто снимал экран и может проверить. (мой модуль пока не пришел).
Спасибо.

p.s. шить модуль планирую с помощью sblAppEx.exe (swra466c) с подменой бинарника.

egony, 2021-03-19:
Bootloader на DIO_15

closed 2021-03-19 by egony

wileu, 2021-03-22:
I bought E72-2G4M20S1E on Aliexpress. How can I flash this module? UART? How to do this? It is possible first flash through the UART?

wisesokol, 2021-03-22:
I tried, but i hasn’t result. Bootloader pin is dio-13.
Perhaps chip is clean or have other firmware.