egony/cc2652p_E72-2G4M20S1E #15: Poor signal?

sgrunf, 2021-03-26:
Hi, 
i realized this board: https://github.com/egony/cc2652p_E72-2G4M20S1E/wiki/Home-EN#board-for-soldering-sma-directly-to-the-module
I made 5 pieces of it (bought modules on Aliexpress): it works really well, fast and responsive (tested both with your firmware and with koenkk firmware). The problem is that i was expecting more coverage from this chip (i used classic 11 cm antennas). Sensitivity is not very good, i made many test with some xiaomi zigbee sensor and Sonoff devices at long distance or with obstacles: I have to say that result is similar (or even worse) than a cheap cc2531 in the same exact conditions. In some case can't reach a far device that is seen by the 2531.
I must exclude assembly issues because i'm an electronic technician and i design and/or assemble hundreds of boards both for work and in my free time.
Did you ever compared directly real coverage with other adapters?

bulolo, 2021-04-03:
Poor signal！！！！！！！！

egony, 2021-04-04:
> Did you ever compared directly real coverage with other adapters?

Yes.  Coverage on 265p is much better than 2538+2592.

closed 2021-04-04 by egony