egony/cc2652p_E72-2G4M20S1E #16: Firmware question

Gio-dot, 2021-04-08:
Hi Alex, congratulations for your work.
I build my own zigbee adapter using E72-2G4M20S1E chip.
I would like to ask you what are the differences between your latest coordinator firmware and the latest Koenkk one (https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_other_coordinator_20210120.zip).
I'm asking because i constantly find differences in radio performance between the two. Koenkk version give me better performance with sensors that have very low lqi. I tried it several times and the result is always the same. I even found that a far sensor that i use for test is seen with kk firmware (lqi= 0) but is unreachable with your latest version. 
Did you change some radio setting in your last version respect to the Koenkk version?
Thank you.

egony, 2021-04-08:
Hi!
Did you read my firmware description?
My firmware uses default tx power 20, but Koen's - 5. So, my firmware used power amplifier circuits, but Koen's - not used.
Try to set tx power to 5 on my firmware and test on same conditions.

Gio-dot, 2021-04-08:
Today i spent some more hour to repeat again the firmware comparison in this way:

**Hardware**
Usb Zigbee adapter built with E72-2G4M20S1E chip

**Reference firmware**
https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_other_coordinator_20210120.zip

**Compared firmwares**

- https://github.com/egony/cc2652p_E72-2G4M20S1E/blob/master/firmware/coordinator/CC2652P_E72_20210319.zip
- https://github.com/egony/cc2652p_E72-2G4M20S1E/blob/master/firmware/coordinator/CC2652P_E72_20210211.zip

**Conditions**

- Zigbee2mqtt clean install with only one sensor joined (a door/window sensor Xiaomi Mija with 100% battery)
- 4 measure points: _short distance_ (lqi > 100), _medium distance_ (lqi ~ 50), _long distance_ (lqi < 5), _very long distance_ (lqi ~ 0, device become to be unresponsive). During test i lay the sensor on a surface to avoid to touch it and move the magnet to mimic open/close.
- Repeated tests at +5dB and +20dB.

**Conclusions**

I repeated this test dozens of times in different days and hours, the results are always the same: 

-  _short_ and _medium distance_ : all firmwares are similar. 
-  _long distance_: refence firmware always works (some little delay sometimes), compared firmwares 80% of the times doesn't work at all.
- _very long distance_: reference firmware works most of the times (sometimes lost some opening/closing send signal), compared firmwares don't work at all.

Both compared firmwares behaviour is similar, i can't see differences. The test are similar even changing tx power from +5dB to +20dB (i tink tx power doesn't affect receiver sensitivity).

I have no idea why i have different results with your firmwares: i noticed you used new sdk: maybe something is changed in default radio settings? Transmission data rate for example?

Gio-dot, 2021-04-08:
If you have in mind that there can be some radio related difference between your firmwares and the Koen version, i'm available to repeat tests if you give me some modified firmware to test.

Gio-dot, 2021-04-08:
Ok, i can't help to investigate at firmware level, i have no skills on it, but i can test all you want if you find something to tune.

egony, 2021-04-11:
> i'm available to repeat tests if you give me some modified firmware to test.

Hi!
Here 2 firmwares for testing. First recompiled for cc1352p chip variant. Second is the same, but with disabled dcdc.
Please test it.   
https://github.com/egony/cc2652p_E72-2G4M20S1E/tree/master/firmware/coordinator/Testing

Gio-dot, 2021-04-23:
Hi, sorry for the delay, i was very busy. In this days i repeated tests with all firmware (and even with an rf-star adapter). Strangely in this latest comparisons i didn't find the same differences between firmwares as in the past. I found that all have pretty the same behaviour. Conclusion is that is very hard to compare empirically radio performances. I obtained different results in different days. I think this can be related in some environmental interference issue.

egony, 2021-04-23:
Hi!
Thank you for testing!
I tested firmwares too and see no any diference (but yes, different results in different days too).

onlinegill, 2021-04-27:
Hi, I am open to test anything. i have couple PCB boards and I need to place couple component on right place. if you could hep me that would be great. its been couple months I am playing around

closed 2021-04-27 by egony