egony/cc2652p_E72-2G4M20S1E #24: zigbee-herdsman supportsLED patch no longer needed

matbech, 2021-08-04:
The patch described here:
https://github.com/egony/cc2652p_E72-2G4M20S1E/tree/master/firmware/coordinator

```
Leds can be turned OFF by zigbee2mqtt config, but on March 2021 you need to modify file zStackAdapter.js:

supportsLED() {
    return __awaiter(this, void 0, void 0, function* () {
      return true; //this.version.product !== tstype_1.ZnpVersion.zStack3x0; // <---- change like this
    });
```

Doesn't seem to be necessary anymore with the latest version of zigbee-herdsman:
https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/z-stack/adapter/zStackAdapter.ts#L116

egony, 2021-08-05:
Hi!
Yes, true. But my firmware siill  <20210430 :)))
So I must update firmware first.

matbech, 2021-08-05:
Oh that would be much appreciated :-)

Symbol84, 2021-08-21:
Green LED does not work after update 20210813.

egony, 2021-08-22:
> Green LED does not work after update 20210813. Turned on and off "Disable led" nothing changes

LEDs fails on some installations (e.g. on Windows all LEDs works fine). Why? I do not know, so now firmware 20210813 removed.

egony, 2021-09-09:
LEDs fails becuse z2m support too bad. New patch instruction added.

closed 2021-09-09 by egony