egony/cc2652p_E72-2G4M20S1E #25: Problem with clearing the coordinator memory

seventhsite, 2021-09-06:
My z2m can't start

`error 2021-09-06 13:39:54: Error while starting zigbee-herdsman`

I tried to update stick firmware. I have CC2652P E72 2G4M20S1E.  Looks like the update was success:

```
(zigbee2mqtt) homeassistant@raspberrypi:~ $ python3.8 cc2538-bsl.py -p /dev/ttyUSB0 -ewv znp_CC2652P_E72_sdk_5_20_00_52_20210901.hex
Invoking bootloader by method 1 (default)
Opening port /dev/ttyUSB0, baud 500000
Reading data from znp_CC2652P_E72_sdk_5_20_00_52_20210901.hex
Your firmware looks like an Intel Hex file
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:22:AB:78:56
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F980
    Write done
Verifying by comparing CRC32 calculations.
    Verified (match: 0x39d15b77)
```

But something wrong with last step:

```
root@raspberrypi:/opt/zigbee2mqtt/scripts# source /opt/zigbee2mqtt/bin/activate
(zigbee2mqtt) root@raspberrypi:/opt/zigbee2mqtt/scripts# node zStackEraseAllNvMem.js /dev/ttyUSB0
Detected znp version 'zStack3x0' ({"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20210901})
Clearing all NVMEM items, from 0 to 815
NVMEM item #4 - deleting, size: 20
(node:8106) UnhandledPromiseRejectionWarning: Error: SRSP - SYS - nvDelete after 0,9ms
    at Timeout.waiter.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
(node:8106) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:8106) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```
What can I do?

bruvv, 2021-09-08:
same error here

egony, 2021-09-09:
zStackEraseAllNvMem.js fails because something changed in z2m. Wait for fix.

closed 2021-09-09 by egony