egony/cc2652p_E72-2G4M20S1E #26: [SUGGESTION] Switch to USB-to-UART bridge chip with programmable EEPROM so can add custom "Product Description String" string as unique identifier

Hedda, 2021-09-23:
Please consider switching the design to a other USB-to-Serial converter chip with its own writable EEPROM as a hardware feature.

This would allow you to add your own custom "Product Description String" as a unique identifier for it via the USB interface, and the reason for wanting a unique identifier via the USB interface is in order to enable the possibility for it to support automatic USB discovery of Zigbee USB adapters. 

The whole point of this is to make it possible for developers to make the initial installation of Zigbee solution plug-and-play friendly and easier for different home automation software to automatically USB discover and initiate a setup without end-user interactions.

Support for USB discovery was recently added to Home Assistant OS (formerly HASSIO) and the ZHA integration for it, see here:

https://community.home-assistant.io/t/unique-friendly-name-description-for-automatic-zigbee-usb-adapter-discovery-in-home-assistant-zha-using-dongle-vendor-product-ids/337077

That unique customized USB description string could be something like ex.; "_**Egony Ebyte CC2652P Zigbee stick version 5 hardware revision 5.0.0**_" and/or "_**Egony Ebyte E72-2G4M20S1E CC2652P Zigbee stick version 5 hardware revision 5.0.0**_"

The unique "description" string for each USB adapter can then be added to HA via a PR like this -> https://github.com/home-assistant/core/pull/56201

As I understand it, cheaper CH340 series (example CH340C and CH340E) by WCH which unfortunately does not support this feature.

I understand that more expensive chips like FT231 chips by FTDI and CP210x chips by Silicon Labs / Silabs do support this feature:

https://ftdichip.com/products/ft231xs/
https://ftdichip.com/products/ft231xq/

FT231X/FT231XS: "**_Key Hardware Features_**" "**_Fully integrated 2048 byte EEPROM for storing device descriptors and CBUS I/O configuration._**"

https://www.silabs.com/interface/usb-bridges/classic/device.cp2102
https://www.silabs.com/interface/usb-bridges/classic/device.cp2104
https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers
https://www.silabs.com/documents/public/data-sheets/cp2102n-datasheet.pdf
https://www.silabs.com/documents/public/application-notes/an197.pdf
https://www.silabs.com/documents/public/application-notes/an978-cp210x-usb-to-uart-api-specification.pdf
https://www.silabs.com/documents/public/application-notes/AN571.pdf

"**_The CP2102N devices have the following features_**" "**_Internal 960-byte programmable ROM for vendor ID, product ID, serial number, power descriptor, release number, and product description strings_**"

"**_The CP2102N includes an internal electrically erasable programmable read-only memory (EEPROM). This memory may be used to customize the USB Vendor ID (VID), Product ID (PID), Product Description String, Power Descriptor, Device Release Number and Device Serial Number as desired for OEM applications. If the EEPROM is not programmed with OEM data, the default configuration data shown in the table below is used._**"

"**_Product String The Product String is an optional string that describes the product. It is limited to 126 characters._**"

PS: I understand that a bonus feature in of chips like FT231 and CP210x is to add the ability to allow users to auto-reset and put the device in bootloader mode via the DTR/RTS pins exposed will enable much easier firmware upgrades for the end-users as they will no longer need to press any buttons to enter bootloader mode, and then bootloader mode could be activated via software from the firmware flasher program software which could make the OTW (Over-The-Wire) firmware upgrade procedure more user-friendly.

Hedda, 2021-10-05:
@egony FYI; I noticed this PR to Home Assistant Core which indicates that tube0013 must have managed have added USB discovery in Home Assistant and ZHA for tubeszb ch340B based serial devices by configuring a unique "Product Description String":

https://github.com/home-assistant/core/pull/56719

So the CH340B is another USB to serial bridge chip that integrates EEPROM used to configure the "Product Description String", etc.

http://www.wch-ic.com/products/CH340.html

Hedda, 2021-10-06:
Note! Home Assistant and ZHA uses the descriptor in combination with VID and PID to auto detect and identify unique devices.

You do not want to change VID and PID as the operating system uses those two to select which device driver to use.

This way multiple devices from different manufacturers can have same VID and PID while still being detected by app as unique.

Warning! If you do change the VID or PID you must provide the customer with your own custom driver for your specific device.

PS: Check out some more discussion in regards to ZHA discovery here -> https://github.com/zigpy/zigpy/discussions/808

egony, 2021-12-25:
CH40B accepts only 18 chars in product string. Too expensive, too much time to program chip, too pointless. 
BTW, I not use HA and never planed.