egony/cc2652p_E72-2G4M20S1E #27: MEM_ERROR since FW update to 20210901

ALaDoffe, 2021-09-26:
## What happened

Hello 

I recently updated my cc2652p_E72-2G4M20S1E  firmware from 20210211 to 20210901
I followed the procédure https://github.com/egony/cc2652p_E72-2G4M20S1E/wiki/Flashing-EN 

Flash was sucessfull and erased NVRAM at the end as described.
  
Since I got regulary 0x10: MEM_ERROR and Data request failed with error: 'MAC no resources' (26)

Need to unplug/plug the stick and work for a while. 

I have only 28 devices and most of them are routers.

## What did you expect to happen

Work as before

## Debug info

Zigbee2MQTT version: Zigbee2MQTT version 1.21.1-dev (commit #58957a30) on RPI 3B+ Venv install
Adapter hardware: CC2652P E72-2G4M20S1E
Adapter firmware version: CC2652P_E72_20210901
log : [zigbee2mqtt.zip](https://github.com/egony/cc2652p_E72-2G4M20S1E/files/7230726/zigbee2mqtt.zip)

ALaDoffe, 2021-09-28:
Another issue with the latest firmware https://github.com/Koenkk/zigbee2mqtt/issues/8870

anyuta1166, 2021-09-30:
I have the same problem with RF-STAR based stick: https://github.com/egony/cc2652p_cc1352p_RF-STAR/issues/2
zigbee2mqtt crashes every few hours with 0x10: MEM_ERROR
According to koenkk, this is a known issue with SDK 5.20.00.52: https://github.com/Koenkk/zigbee2mqtt/issues/8885#issuecomment-929980617

egony, 2021-12-25:
Yes, is a known issue with SDK 5.20.00.52

closed 2021-12-25 by egony