egony/cc2652p_E72-2G4M20S1E #33: Zigbee2MQTT:error - Failed to set permit join to true

sergey-suloev, 2021-12-25:
Zigbee2MQTT:warn 2021-12-21 12:22:59: permit_join set to true in configuration.yaml.
Zigbee2MQTT:warn 2021-12-21 12:22:59: Allowing new devices to join.
Zigbee2MQTT:warn 2021-12-21 12:22:59: Set permit_join to false once you joined all devices.
Zigbee2MQTT:info 2021-12-21 12:22:59: Zigbee: allowing new devices to join.
Zigbee2MQTT:error 2021-12-21 12:22:59: Failed to set permit join to true

sergey-suloev, 2021-12-25:
This is not a fake but a real issue

sergey-suloev, 2021-12-25:
Used firmware CC2652P_E72_20210319

egony, 2021-12-25:
> This is not a fake but a real issue. It happens very often with latest Zigbee2MQTT. So take a look at it.

Why? Failed to set permit join to true in log not means that permit join is not set to true.

bcdonadio, 2021-12-25:
If I may, I think I saw this issue a few times in the last weeks (no logs right now, sorry, but I'll keep an eye). I think it's more likely caused by the GUI of Z2MQTT failing internally to perform this action instead of something to do with the firmware.

There was a supposedly solved bug not so long ago that it failed to do that selectively (just for the controller or for a single router).

bcdonadio, 2021-12-25:
And also IIRC, with ZB3.0, Z2MQTT would change your permit_join to false after starting anyway.

sergey-suloev, 2021-12-25:
> If I may, I think I saw this issue a few times in the last weeks (no logs right now, sorry, but I'll keep an eye). I think it's more likely caused by the GUI of Z2MQTT failing internally to perform [...]

if you are sure that it is not a firmware bug then the issue can be closed

closed 2022-02-15 by egony